franfrandev/OctoPrint-CalibrationCompanion #1: Installation link broken

adam5zki, 2021-05-11:
The installation via the Plugin manager fails due to a HTTP404 error. The link seems to be wrong as it does report a 404 error. The link in the description's Installation section "https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/archive/master.zip" throws the same 404 error.

dualznz, 2021-05-11:
i can confirm this error as well

jasonmacdonald, 2021-05-11:
Someone beat me to it. It's broken! :)

Gruvyminion, 2021-05-11:
Same:

> Installing plugin "Calibration Companion" from https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/archive/master.zip...
Error!
Could not fetch plugin from server, got 404 Client Error: Not Found for url: https://codeload.github.com/iFrostizz/OctoPrint-CalibrationCompanion/zip/master

CaptClaude, 2021-05-11:
That fixed it for me! Thanks.

closed 2021-05-11 by iFrostizz