franfrandev/OctoPrint-CalibrationCompanion #6: QA/UAT Issue - Progress Bar and Cancel buttons - PID autotune tab

agentchimendez, 2021-05-13:
Opening ticket for the PID autotune tab but the issue may effect more aspects of the plug in. I will expand this issue or create a new one as I test the plug in.

Expected Behaviour

After clicking button to start PID autotune('Run PID AutoTune') a progress bar and cancel button should appear.

Current Behaviour

After clicking button to start PID autotune no indication is given that process has started or a means to cancel the process.

Side-Effects

Due to the PID Autotune starting the printer does not respond to other commands. The fan toggles in the PID Autotune tab become unresponsive. If a user leaves the Calibration-Companion window and attempts commands from the Octopi 'Control' tab the printer will not respond. A user can open the 'Terminal' tab and see in the console output that the PID autotune is running(filtering out temperature messages helps).

Possible Bug

While the printer is unresponsive it does recognize and store the entered commands to its buffer. These commands are successfully run after the PID autotune is complete. This presents two problems.

A) Commands may be issued in such an order that damage occurs to the printer. Through some combination of commands clicked it is conceivable to induce thermal issues or crash the nozzle to the build plate.

B) If a user continues clicking buttons it may overflow the buffer. I have no idea what would happen in this case but I assume the mainboard would crash. This could present very confusing behaviour to a user more familiar with the hardware end of printing than the software. It could present a security issue as well but it doesn't appear to be a practical threat vector.

Notes

This functionality may be limited by Marlin features. If that is the case a disclaimer stating that process cannot be stopped and a second confirmation to start would be a good backstop to counter expectations set by general computer use. 

If progress cannot be shown(or is impractical due to the variety of printers) showing relevant console output in the U.I can serve as an indication that something is happening. Most users of Octopi are familiar with M/G commands and their console outputs, as such a console output does not present the same barrier to entry to users as it normally would.

Platform:

Anycubic i3 Mega S
Marlin 2.x/ Knutwurst 1.1.9
Octopi - Raspi 4 4GB
BLTouch 3.1

iFrostizz, 2021-05-13:
Hello,
First, thank you very much for this very detailed explanation.
The PID Autotune is blocking, which means that you cannot do anything while it is running.
As you mentioned it, the commands are getting stored and running one after one.
A nozzle crash would totally be possible, so I'll implement a progress bar as you mentionned it using the PID cycles, but hopefully you will not crash the mainboard because of buffer overflow.
Thanks again!

iFrostizz, 2021-05-14:
Hello,
I prepared for you a progress bar on the PID autotune as well as some minor bug fixes / functionnalities.
[OctoPrint-CalibrationCompanion.zip](https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/files/6478624/OctoPrint-CalibrationCompanion.zip) if you feel like, you can download it from the plugin manager "from an uploaded file".
They are some chances for the progress bar to not work as I don't have any printer right now that I can use, I'm not at home.
I based myself on the received commands from Marlin after a PID: 
[image]
Taken here: https://support.th3dstudio.com/hc/guides/diy-guides/p-i-d-hotend-calibration-guide/
I also added the possibility for the user to choose the number of cycles of PID Autotune. So each time that it receives this kind of line: 
[image], it should iterate until reaching the number of cycles, and I also removed the `M500` saving command for the test purpose.
Hope that it's gonna work !
Good day ;)

iFrostizz, 2021-05-17:
I just added an alert message explaining briefly that it is not a good idea to flood the buffer by sending commands while running the PID autotune for the newer version. The progress bar will be for later, I need to test it first.

iFrostizz, 2021-06-20:
Fixed in 0.1.1

closed 2021-06-20 by iFrostizz