franfrandev/OctoPrint-CalibrationCompanion #7: CalibrationCompanion breaks Spoolmanager

GylRon1, 2021-05-13:
After installing CalibrationCompanion it is impossible to add new spools in Spoolmanager. If it is disabled and Octoprint is restarted, adding spools is possible again.  I have Themeify installed. Attached an empty spool adding screen grab (the error) and a correct working screen grab.
[image]
[image]

iFrostizz, 2021-05-13:
Do you have any error in the devtools (F12) when calibration companion is enabled ? Usually this is a conflict between Ids, or some kind of knockout bindings

GylRon1, 2021-05-13:
Sorry, I'm not the smartest when it comes to programming, I don't know where to find the devtools (F12). I'm working with the Chrome browser if that can help?

iFrostizz, 2021-05-13:
Just enable Calibration Companion, go in the SpoolManager tab (where it's broken), and press "F12". Here, a tab should open, and head on "Console". There might be some errors in red. If so, please provide me a screenshot

GylRon1, 2021-05-13:
[image]
Here you go.

iFrostizz, 2021-05-14:
I don't think that this is the error I'm searching for, I see that they are 33 errors 
[image]
Do you spot anything else ?

GylRon1, 2021-05-14:
Hello iFrostizz, the other errors were network related. I prepared another Raspberry Pi (4B 8GB) with latest octopi and newest Octoprint with the same plugins and notice the same problem. I don't have a printer attached, but tested on my regular Octoprint that it does not matter for the error to occure. I have attached the console output on the fresh install. There are no other errors, only 6 warnings. Hope this helps.
[image]

iFrostizz, 2021-05-14:
Thanks, it was a css from my plugin. I'm gonna publish an update soon

GylRon1, 2021-05-14:
Thanks for looking in to it. I appreciate all you guys for the work you do and making this all possible for us (the lesser gifted in programming)!

closed 2021-05-17 by iFrostizz

GylRon1, 2021-05-17:
Nice, Spoolmanager seems to work properly again after your correction.

Keep up the good work,

Best regards,
Ronny.