franfrandev/OctoPrint-CalibrationCompanion #9: Bed leveling tab does nothing

A-Swafford, 2021-05-14:
Using an ender3 pro and latest octoprint. The profiles save and load correctly but clicking buttons on tabs other than "save" or "load" does nothing.

Attempted to use bed leveling, PID tune, and print squares. All did nothing.

iFrostizz, 2021-05-14:
Do you have any files in the file manager ?

A-Swafford, 2021-05-14:
It totally did add those files, I missed the part in the instructions where I would need to manually start the files going. I interpreted "file would be added to manager to..." meaning it would be auto started (like what prusa does). Thanks!

iFrostizz, 2021-05-14:
I could do that but I prefere to let the control to the user :)

closed 2021-05-14 by iFrostizz

iFrostizz, 2021-06-22:
An "auto-print" feature has been added in the setup tab of the 1.0.1 version