franfrandev/OctoPrint-CalibrationCompanion #12: Profiles load and save dont work

jeanfbrito, 2021-05-18:
Nothing happens when change the profile and use load or save button. Browser console dont show any error.
What I can send to you to help debugging?
[image]

iFrostizz, 2021-05-19:
No problem on my side, I noticed that the settings were taking a lot of time to load on some Octoprint instances.
Maybe that you can see something in the logs ? [https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1](url)

jasc76, 2021-05-19:
Same issue for me. Yes loading takes some time. But it doesnt store

iFrostizz, 2021-05-21:
So it seems like manipulating all these settings is very resource consuming. They do work but are incredibly slow, I sadly coudn't try it on a pi... I will find a solution hopefully.

jeanfbrito, 2021-05-21:
If you serialize the settings into a JSON only?

iFrostizz, 2021-05-22:
@jeanfbrito The solution was to get the settings from the settingViewModel rather than call them from the config.yaml file, which is doing API calls. @cp2004 put me on the way :)
I have made a "devel" branch for testing purpose and this one seems to work better.
https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/tree/devel

iFrostizz, 2021-06-20:
Fixed in 0.1.1

closed 2021-06-20 by iFrostizz

csprocket777, 2021-07-17:
I'm on 1.0.1, still having this issue where no settings will save. The save button will act like it's complete but when you click the "load" button, the default settings reappear. @iFrostizz

iFrostizz, 2021-07-17:
> I'm on 1.0.1, still having this issue where no settings will save. The save button will act like it's complete but when you click the "load" button, the default settings reappear. @iFrostizz

With the 1.0.1, I added a logging message in the console tab of the "devtools" in the browser.
[image]
Do you have such a message ? Alternatively, some errors could appear and your settings would not be saved.

csprocket777, 2021-07-27:
Ok. I'm not sure what changed. This time I hit reset on each profile and then modified Profile 1 and saved and this time it saved. ¯\\\_(ツ)_/¯. Please close.