franfrandev/OctoPrint-CalibrationCompanion #13: Big retraction at the start of temp tower

moorsey, 2021-05-20:
When printing a temperature tower, before the print starts, there is a big retraction, meaning the print fails

Not that good with the g-code, nothing obvious I can see that would be causing it

Logs from Octoprint Terminal tab below, ran from clicking start, to cancelling just before first layer finished.  Gcocde generated (first layer) at the bottom also

```
Recv:  T:205.16 /205.00 B:69.83 /70.00 @:60 B@:0
Recv:  T:205.00 /205.00 B:69.81 /70.00 @:63 B@:127
Recv:  T:205.00 /205.00 B:70.08 /70.00 @:63 B@:127
Recv:  T:205.00 /205.00 B:70.33 /70.00 @:63 B@:0
Recv:  T:205.04 /205.00 B:70.33 /70.00 @:62 B@:0
Recv:  T:205.00 /205.00 B:70.28 /70.00 @:63 B@:0
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 G28*18
Recv:  T:205.00 /205.00 B:70.05 /70.00 @:63 B@:0
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:69.85 /70.00 @:63 B@:0
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:69.84 /70.00 @:63 B@:127
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:70.10 /70.00 @:63 B@:127
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:70.34 /70.00 @:63 B@:0
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:70.34 /70.00 @:63 B@:0
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:70.32 /70.00 @:63 B@:0
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:70.16 /70.00 @:63 B@:0
Recv: echo:busy: processing
Recv:  T:205.04 /205.00 B:70.00 /70.00 @:62 B@:0
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:69.83 /70.00 @:63 B@:127
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:70.00 /70.00 @:63 B@:127
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:70.20 /70.00 @:63 B@:127
Recv: echo:busy: processing
Recv:  T:205.16 /205.00 B:70.43 /70.00 @:59 B@:0
Recv: echo:busy: processing
Recv:  T:205.31 /205.00 B:70.40 /70.00 @:56 B@:0
Recv: echo:busy: processing
Recv:  T:205.20 /205.00 B:70.32 /70.00 @:59 B@:0
Recv: echo:busy: processing
Recv:  T:205.04 /205.00 B:70.15 /70.00 @:62 B@:0
Recv: echo:busy: processing
Recv:  T:205.00 /205.00 B:69.99 /70.00 @:63 B@:0
Recv: X:157.00 Y:117.00 Z:11.43 E:21.82 Count X:12617 Y:9435 Z:9214
Recv: ok
Send: N2 M117 ETA 5/20 10.58PM*113
Recv: ok
Send: N3 M73 P0*84
Recv: echo:Unknown command: "M73 P0"
Recv: ok
Send: N4 M117 Good accuracy*30
Recv: ok
Send: N5 M73 P0*82
Recv: echo:Unknown command: "M73 P0"
Recv: ok
Send: N6 M117 0.00% complete*16
Recv: ok
Send: N7 M73 P0*80
Recv: echo:Unknown command: "M73 P0"
Recv: ok
Send: N8 M420 S1*111
Recv: echo:Bed Leveling ON
[... 605 lines cut ...]
G0 F9000 X120.548 Y109.275;
G1 F1500 X124.723 Y113.450 E9.63807;
G0 F9000 X125.289 Y113.450;
G1 F1500 X120.548 Y108.709 E9.86107;
G0 F9000 X120.954 Y108.549;
G1 F1500 X125.449 Y113.044 E10.07250;
G0 F9000 X125.449 Y112.478;
G1 F1500 X121.520 Y108.549 E10.25731;
G0 F9000 X122.086 Y108.549;
G1 F1500 X125.449 Y111.912 E10.41550;
G0 F9000 X125.449 Y111.347;
G1 F1500 X122.651 Y108.549 E10.54711;
G0 F9000 X123.217 Y108.549;
G1 F1500 X125.449 Y110.781 E10.65210;
G0 F9000 X125.449 Y110.215;
G1 F1500 X123.783 Y108.549 E10.73046;
G0 F9000 X124.348 Y108.549;
G1 F1500 X125.449 Y109.650 E10.78225;
G0 F9000 X125.449 Y109.084;
G1 F1500 X124.914 Y108.549 E10.80741;
G1 F4200 E5.80741;
G0 F9000 X124.914 Y108.549 Z0.40;
G0 F9000 X94.000 Y114.000;
M104 S190;
```

iFrostizz, 2021-05-20:
Hello,
You may need to add a `M90 E0` in your start gcode https://marlinfw.org/docs/gcode/G092.html. I added it in the new version of the plugin but your settings are saved through versions.

moorsey, 2021-05-22:
Hey thanks, added that at the end of the start gcode box, but no dice, but might have it in the wrong order I guess!

iFrostizz, 2021-05-22:
Can you please send me your .gcode file ?

iFrostizz, 2021-05-26:
@moorsey Sorry, I sent you the wrong command but the right link lol...
That's `G92 E0` to reset the extruder.
The end of the start gcode was the right place.

closed 2021-05-28 by iFrostizz

moorsey, 2021-06-10:
Apologies, the alternative command does do the trick, sorry for not replying, many thanks!