franfrandev/OctoPrint-CalibrationCompanion #15: Web UI initialization takes 1.5 minutes more than without this plug-in enable

Misterke, 2021-05-21:
Please note that I am just reporting this with relatively little info because I couldn't pinpoint any useful logging for it.  I often use an Android app to start / monitor OctoPrint and hence don't always use the web UI.  As such I only noticed that loading it had become very very very slow - or rather: the GUI elements would load normally, but getting them filled with the appropriate data took minutes.

I then enabled logging of events and filemanager for OctoPrint and what I noticed was this in the octoprint.log:

```
2021-05-21 11:49:13,841 - octoprint.plugins.PrintJobHistory - INFO - Plugin-State: PreHeat=enabled DisplayLayerProgress=enabled filamentmanager=enabled ultimakerformat=enabled PrusaSlicerThumbnails=enabled
2021-05-21 11:49:13,844 - octoprint.events.fire - DEBUG - Firing event: UserLoggedIn (Payload: {'username': 'admin'})
2021-05-21 11:50:45,350 - octoprint.server.util.sockjs - INFO - User admin logged in on the socket from client ::ffff:192.168.21.183
2021-05-21 11:50:45,394 - octoprint.events.fire - DEBUG - Firing event: ClientAuthed (Payload: {'username': 'admin', 'remoteAddress': '::ffff:192.168.21.183'})
```

Note that between UserLoggedIn and the ClientAuthed (I have no idea what the difference is - I'm a noob here) there's 1.5 minutes.  Trying to make sense of the network requests in the browser did not tell me much - there were just too many requests quite some of them would hang for a long time.

I then went the other path: disabling all installed 3rd party plugins.  After this the Web UI was lightning fast.  Then I started reenabling sets of plug-ins in order of what I considered increasing likelihood of causing something like this.  When enabling the bunch where this plugin was part of (which I had actually given low chance of causing the problem) the initialization problems of the Web UI came back.  I then started looking for the plug-in in this set that caused the problem to go away or come back and ended up with ... Calibration-Companion.

Now, the easy solution for me is to just disable / remove this plug-in, so that's what I did.  But as people on the OctoPrint discord rightly indicated, better inform the author of the problem I was hitting.  I realize you will likely be unable to proceed with just this limited info and I don't have any logging to supply, so ...  If needed I can collect extra info if you instruct me to what you want me to try, but it will not be my top-priority ...

iFrostizz, 2021-05-21:
Thanks for this issue, it seems like getting the data from the settings is slow the way it is doing it. Same issue as here: https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/issues/12

GylRon1, 2021-05-21:
I can confirm this problem, uploaded filelist takes forever to show on RPi3b gui, the i/o button to reboot the system does not appear, in the Octoprint settings / Access Control no users show, webcam stream does not work until the link is clicked, ... . Only proper solution is to disable the Calibration plugin. After reboot all seems normal again.

iFrostizz, 2021-05-22:
I have made a "devel" branch for testing purpose and this one seems to work better.
https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/tree/devel

GylRon1, 2021-05-23:
Still slow on my system, the i/o button to reboot the system does not appear, in the Octoprint settings / Access Control no users show, no files list, and the webcam stream only appears after clicking the "let's go !" button in your plugin's tab. Pop-up messages do not appear. Diificult to reboot the system, installed plugins do not show, refreshing that list has to be done by first going to the tab to install new plugins and closing it again. After that disabling/uninstalling the plugin is possible. Restart is only possible via SSH. I uninstalled the plugin for now, causes too much inconveniences.

iFrostizz, 2021-05-24:
@GylRon1 I absolutely understand you, but did you tried the one from the devel branch ? Anyway, XAMCTBO is testing the plugin and it looks much faster so far.

GylRon1, 2021-05-24:
Yes, My latest comment was after installing the dev branche and
trying/surveying several things. I guess I better swap my old RPi3b in
favor of a new RPi4.

Best regards,

Ronny.

lsaranto, 2021-05-28:
The devel version (shows 0.1.1rc0) seems to be faster, and ok. At least for me.

iFrostizz, 2021-06-20:
Fixed in 0.1.1

closed 2021-06-20 by iFrostizz