franfrandev/OctoPrint-CalibrationCompanion #17: Print does not start after clicking "print retraction tower"

ventilator, 2021-05-24:
Hi, I am a bit unsure, how to start the generated gcode.
I set up a retraction tower, then hit "Print retraction tower". Then nothing happens.
In the octoprint log file there is logged the gcode, then "Analysis of entry local:voron_ABS_0.4_retra_tower.gcode finished", then nothing more logged.

Y59.769 E356.95636;\nG1 F3600 X75.759 Y59.907 E356.96149;\nG1 F3600 X75.816 Y60.056 E356.96680;\nG1 F3600 X75.863 Y60.238 E356.97305;\nG1 F3600 X75.882 Y60.380 E356.97782;\nG1 F3600 X75.888 Y60.562 E356.98388;\nG1 F3600 X75.877 Y60.699 E356.98845;\nG1 F3600 X75.842 Y60.885 E356.99474;\nG1 F3600 X75.792 Y61.045 E357.00032;\nG1 F3600 X75.724 Y61.202 E357.00601;\nG1 F3600 X75.642 Y61.345 E357.01149;\nG1 F3600 X75.555 Y61.463 E357.01637;\nG1 F3600 X75.437 Y61.594 E357.02223;\nG1 F3600 X75.295 Y61.719 E357.02852;\nG1 F3600 X75.196 Y61.788 E357.03253;\nG1 F3600 X75.013 Y61.887 E357.03945;\nG1 F3600 X74.878 Y61.943 E357.04431;\nG1 F3600 X74.704 Y61.991 E357.05031;\nG1 F3600 X74.559 Y62.015 E357.05520;\nG1 F3600 X74.382 Y62.025 E357.06110;\nG1 F3600 X74.209 Y62.016 E357.06686;\nG1 F3600 X74.057 Y61.991 E357.07198;\nG0 F7800 X74.845 Y61.782;\nG1 F3600 X75.641 Y60.986 E357.10942;\nG0 F7800 X75.714 Y60.347;\nG1 F3600 X74.207 Y61.854 E357.18030;\nG0 F7800 X73.773 Y61.722;\nG1 F3600 X75.582 Y59.914 E357.26537;\nG0 F7800 X75.347 Y59.583;\nG1 F3600 X73.441 Y61.489 E357.35502;\nG0 F7800 X73.198 Y61.166;\nG1 F3600 X75.028 Y59.336 E357.44110;\nG0 F7800 X74.611 Y59.188;\nG1 F3600 X73.052 Y60.747 E357.51443;\nG0 F7800 X73.085 Y60.148;\nG1 F3600 X74.014 Y59.219 E357.55813;\nG0 F7800 X73.900 Y58.889;\n')])])
2021-05-24 18:21:43,912 - octoprint.filemanager.analysis - INFO - Starting analysis of local:voron_ABS_0.4_retra_tower.gcode
2021-05-24 18:21:43,915 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/voron_ABS_0.4_retra_tower.gcode
2021-05-24 18:21:47,180 - octoprint.filemanager.analysis - INFO - Analysis of entry local:voron_ABS_0.4_retra_tower.gcode finished, needed 3.27s

ventilator, 2021-05-24:
Oh, sorry for asking; i figured it out, the code is available on the file system of Octoprint and is not automatically startet

closed 2021-05-24 by ventilator

iFrostizz, 2021-05-28:
The next released version will be more intuitive will loadings, and an "auto-print" feature.