franfrandev/OctoPrint-CalibrationCompanion #18: Acceleration Tower includes moves to a x negative position

ventilator, 2021-05-25:
Hi, 
I am trying to print an acceleration tower on my voron0, which has a limited print bed of only 119mm. Is this the reason, the generated tower gcode includes negative x values? Can I somehow print the tower, e.g. make it smaller?

Issue: "Move out of range: -0.100 30.400 0.200 [15.802]"
Gcode:
;---------START G-CODE---------

G0 F7800 X79.5 Y110;
G0 F7800 Z0.2;
M204 P500;
M205 X6 Y6;
M117 St1/4 Acc500 Jx6Jy6;
;LAYER:0
G1 F1800 X41.032 Y71.532 E1.80942;
G1 F1800 X41.032 Y68.586 E1.90740;
G1 F1800 X38.086 Y68.586 E2.00538;
G1 F1800 **_X-0.100_** Y30.400 E3.80153;  <------here

[image]

iFrostizz, 2021-05-25:
Hello,
The acceleration tower is 120mm wide.
I'm going to make a smaller one in the next release, are you fine with a 100*100mm tower ?

ventilator, 2021-05-28:
(100mm)² would be very nice :-) - my printer is a Voron0, an open source
printer with a bed of (120mm)², but usually only 116..119mm are usable

iFrostizz, 2021-06-20:
Fixed in 0.1.1

closed 2021-06-20 by iFrostizz