franfrandev/OctoPrint-CalibrationCompanion #20: Nozzle sizes greater that 0.8 not supported

jimbo96, 2021-06-07:
Nozzle sizes 0.4-0.8 are supported in CC, but nothing larger. I'm using 1.0 and 1.2 nozzles and would like to see support for these added.

iFrostizz, 2021-06-07:
I will add these in the next update

Neelix96, 2021-06-19:
also 0.3mm nozzles would be nice to have

iFrostizz, 2021-06-20:
Adding the gcode to generate it is not the most fun task and is taking time. So I will first update the plugin with the smaller acceleration tower, some bug fixes and everything that is already in the dev branch.
The new nozzle sizes will come later :)

tyeth, 2021-08-29:
Just to add I'm currently trying to establish my nozzle size so am changing between 0.3, 0.35, 0.4 and 0.5mm, is it possible to have a dynamic input box, maybe generating the gcode via a python script instead of by hand for each variation?

iFrostizz, 2021-08-29:
Sadly no @tyeth ,
I would need to program an entire slicer if so. Because a smaller nozzle would induce more walls, different gcode positions, ...
Probably too complicated for a simple script

tyeth, 2021-09-04:
Thanks for the update. I bought some mini drill bits and found mines 0.5mm