franfrandev/OctoPrint-CalibrationCompanion #24: plugin hangs after clicking 'print nnnnn'

L4RGER, 2021-07-02:
after setting up one of the tests and clicking the 'print xxxxxx' button at the bottom the file is generated and then can be printed but the plug button remains as 'generating.....' and requires a restart to generate the calibration print again.

iFrostizz, 2021-07-02:
It looks like it's working on my side, is that on a specific file, or all ?
Do you have such a message after generating the .gcode ? 
[image]
Do you have any error in the logs ?

L4RGER, 2021-07-02:
Hey! 
yep it creates the file fine. I just did the first layer one again through chrome browser on my phone and it seems fine. 
I was using Firefox on my MacBook earlier, maybe it’s a Firefox issue rather than a plug in issue. 
sorry about opening this unnecessarily

L4RGER, 2021-07-02:
Also thanks for creating this.

closed 2021-07-02 by L4RGER