franfrandev/OctoPrint-CalibrationCompanion #25: Retraction Test Generated Code Includes No Temperature

holocronology, 2021-07-20:
I attempted to generate a gcode print for a Retraction Test. Despite having temperature setting, none were included in the generated gcode.

First part of code: https://pastebin.com/XVWhDNB3

Settings:
[image]

iFrostizz, 2021-07-20:
You can use the variables as described in the "README" file in your start gcode.
You can use the profile 1 as an example.

holocronology, 2021-07-20:
Yes, I have start and end codes. But if you input the temperature settings in the profile, they should just automatically be included in the gcode. Otherwise, why even have these settings if I need to manually code it?

iFrostizz, 2021-07-20:
I merged some changes from the "devel" branch to the "main" branch and the README.md wasn't up to date, now it is.
These are the variables as explained before: https://github.com/iFrostizz/OctoPrint-CalibrationCompanion#variables
I did that to let the user having the freedom to custom its gcode.
Here is an example of my start gcode, please adapt it to your needs.

G90;absolute mode
G0 F[travel_speed] X10 Y10 Z10;place the nozzle for heating
M140 S[regular_bed];set bed to [regular_bed]C
M190 S[regular_bed];wait for bed to [regular_bed]C
M104 S[regular_nozzle];set hotend to [regular_nozzle]C
M109 S[regular_nozzle];wait hotend to [regular_nozzle]C
G0 Z0.3 F[travel_speed];
G1 X10 Y100 E10 F2000;
G92 E0;reset extruder
G0 Y110 Z10 F[travel_speed]

holocronology, 2021-07-20:
OKay, thanks.

iFrostizz, 2021-07-21:
Yeah, try it out :)

closed 2021-07-21 by iFrostizz