franfrandev/OctoPrint-CalibrationCompanion #26: PID tuning both extruder / bed tuning"hangs" after reaching targeted bed temperature

ruedli, 2021-08-17:
Tried this 3 times now, on Prusa MK3S / MMU2S latest firmware.
Plugin version 1.01
octoprint 1.7.0rc2

The extruder temperature seems to go though the tuning process, but the BED tuning never ends. I waited more than 25 minutes, no indication this will finish any time soon.

[image]

Let me know if further information is needed.

ruedli, 2021-08-17:
Additional information: when I only do a bed tuning, it behaves the same: hangs after the BED tuning at the targeted temperature. See attached log of what I see in the serial monitor.

[bed_tune.zip](https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/files/7000366/bed_tune.zip)

iFrostizz, 2021-08-17:
Hi, why is the bed tuning "hanging" ? Does it seems to never end on the plugin interface, or on your printer screen ?
On the logs side, it looks good: the 5 cycles are done and finishes with `Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h`
The only thing is that the bed temperature seems to stay at 60°C after that the autotune has finished.

ruedli, 2021-08-17:
True, the bed temperature stays at 60°C, and there is no indication (other than the line in the commandlog, that disappears when not carefully watched that tuning is "over". Your comment on the plugin interface, made me look more into that. Until now I had not noticed the slightly darker progress line, as it was not updated right after the tuning is inititated. It started like this, 

[image]

Then went in chunks of 30% ? To this: 

[image]

Where it stayed forever... So maybe a good idea is to set the temperature to 0 (stop heating) and displaying the text ready somewhere as well as showing a percentage complete. Just the bar (that only changes after quite a long wait) is easy to oversee.

I initially closed the dialog and switched to the temperature graph to see what was going (sorry for the misunderstanding). If you display the temperature, as well as the calculated PID values, the user can easily see that is finished, this is where I failed.

Thanks for your swift response and looking into it! You can close the issue if you prefer -or- close it after adding some information of the progress (and setting the temperature to 0 after it is at 100%.

iFrostizz, 2021-08-18:
Indeed, the progress bar is staying at 100% when the PID process is finished, and should go back to 0 instead. The only time when it goes at 0 is when a new PID autotune is started.
But I don't understand how the bed temperature could stay at 60°C as if a "M190 ....." was sent to the printer, which the plugin is not doing. The only `M190` that is in the code is in the start gcode, which is not triggered in this case. I'm not sure if this is a side effect of the M303 command, but if it's the case I will add the command to stop the bed heating.
Your idea of displaying the PID constants is good, I'll do that for the next update as a notification at the end of the PID autotune.

ruedli, 2021-08-21:
@iFrostizz 

> But I don't understand how the bed temperature could stay at 60°C as if a "M190 ....." was sent to the printer

This could be how the Prusa firmware works. I tried to follow it: in Marlin_main.cpp they parse the M303 command and (just) call PID_autotune. This is in temperature.cpp and ends like this:

    if(pid_cycle > ncycles) {
      SERIAL_PROTOCOLLNPGM("PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h");
	  pid_tuning_finished = true;
	  pid_cycle = 0;
      return;
    }
    lcd_update(0);
  }
}

As far as I can see they do not reset the target temperature to 0, so maybe that is the problem?

iFrostizz, 2021-08-21:
@ruedli 
That's a very good find, so it looks like the `pid_tuning_finished` boolean is set to `true` after that the PID tuning has been finished. I found that then: https://github.com/prusa3d/Prusa-Firmware/blob/538ce06bf036eb45b0b3bbef6fa50bd694f0c3a3/Firmware/ultralcd.cpp#L1330
Looks like the hotend(s) are all set to 0 °C in this line:

> `setAllTargetHotends(0);  // reset all hotends temperature including the number displayed on the main screen` 
but the:
> `setTargetBed(0);` 

is indeed not called. So this sentence from you: 

> However it ALSO applies to extruder heating, not just bed heating

makes me doubt now...
If it is the case, maybe that this is either not the place where the PID is finished in the code, or an error in the code...
Maybe that it's a choice from the Prusa devs ? In this case, I don't really see why they have done that. If maybe you can confirm that the hotend is shut down, I will open an issue on their repo linking this issue. If not, I will simply set hotends and bed to 0°C in my plugin. I don't think that it would cause any UX issue.

Do you have the "PID cal. finished" message writter on your screen after the process by the way ?

ruedli, 2021-08-21:
@iFrostizz I can test again tomorrow, but the fact that it sits in the lcd module makes me wonder. Either bad programming, or they are indeed just talking about targets set to display on the lcd. I mean: if you need to program temperature stuff, why rely on a call in the module that does lcd stuff? ultralcd did not come from marlin I believe, so there is potential for error here. ;-) After testing I can tell you more...

ruedli, 2021-08-22:
@iFrostizz 
Sooooooo.... it is confirmed: Extruder stays hot after PID calibration (as well) on a Prusa MK3. The good news is that Prusa are not doing "bad programming", resetting the target temperature in the LCD driving software is what I would call a bad practice! The bad news is that you might consider it a bug. It is not really defined what the exit criteria for M303 are, so I guess we have to deal with different implementations. The call you captured seems just to affect the display of values, to get in the mode of normal display of temperatures on the LCD.

I opened PID calibration for _just_ the extruder and watched the temperature / serial communication in another octoprint browser tab.

[image]

You see the temperature rise after the M303, but target temp remains at zero. Then, just when it has done its 5 swings, it processes the PID stores them and THEN.... the target raises to the M303 temperature. This is because of the temperature messages it receives.

The serial log nicely illustrates this:

Recv: T:21.3 /0.0 B:21.3 /0.0 T0:21.3 /0.0 @:0 B@:0 P:22.8 A:26.2
Send: M303 E0 C5 S210
Recv: PID Autotune start
Recv: T:21.69 @:127
Recv: T:23.97 @:127
Recv: T:28.25 @:127
Recv: T:34.14 @:127
Recv: T:40.34 @:127
Recv: T:46.77 @:127
Recv: T:53.43 @:127
Recv: T:59.45 @:127
Recv: T:65.83 @:127
Recv: T:72.38 @:127
Recv: T:77.96 @:127
Recv: T:84.17 @:127
Recv: T:90.26 @:127
Recv: T:95.42 @:127
Recv: T:101.26 @:127
Recv: T:106.46 @:127
Recv: T:112.00 @:127
Recv: T:116.65 @:127
Recv: T:120.34 @:127
Recv: T:125.47 @:127
Recv: T:130.14 @:127
Recv: T:133.99 @:127
Recv: T:140.78 @:127
Recv: T:146.97 @:127
Recv: T:151.95 @:127
Recv: T:157.93 @:127
Recv: T:163.51 @:127
Recv: T:169.23 @:127
Recv: T:174.98 @:127
Recv: T:179.58 @:127
Recv: T:185.03 @:127
Recv: T:189.86 @:127
Recv: T:194.24 @:127
Recv: T:198.85 @:127
Recv: T:203.79 @:127
Recv: T:208.35 @:127
Recv: T:213.33 @:0
Recv: T:216.72 @:0
Recv: T:218.18 @:0
Recv: T:218.13 @:0
Recv: T:217.40 @:0
Recv: T:216.35 @:0
Recv: T:215.00 @:0
Recv: T:213.39 @:0
Recv: T:211.67 @:0
Recv: T:209.69 @:127
Recv: T:208.35 @:127
Recv: T:208.35 @:127
Recv: T:209.82 @:127
Recv: T:213.23 @:0
Recv: T:215.73 @:0
Recv: T:216.30 @:0
Recv: T:216.20 @:0
Recv: T:214.84 @:0
Recv: T:213.91 @:0
Recv: T:212.14 @:0
[... 77 lines cut ...]
Recv: T:201.7 /210.0 B:21.4 /0.0 T0:201.7 /210.0 @:103 B@:0 P:24.6 A:28.7
Recv: T:205.4 /210.0 B:21.3 /0.0 T0:205.4 /210.0 @:64 B@:0 P:24.6 A:28.9
Recv: T:209.2 /210.0 B:21.6 /0.0 T0:209.2 /210.0 @:23 B@:0 P:24.6 A:28.9
Recv: T:212.1 /210.0 B:21.6 /0.0 T0:212.1 /210.0 @:0 B@:0 P:24.6 A:28.8
Recv: T:213.5 /210.0 B:21.6 /0.0 T0:213.5 /210.0 @:0 B@:0 P:24.6 A:28.6
Recv: T:213.3 /210.0 B:21.4 /0.0 T0:213.3 /210.0 @:9 B@:0 P:24.7 A:29.0
Recv: T:212.3 /210.0 B:21.3 /0.0 T0:212.3 /210.0 @:31 B@:0 P:24.7 A:28.7
Recv: T:211.0 /210.0 B:21.6 /0.0 T0:211.0 /210.0 @:55 B@:0 P:24.7 A:28.8
Recv: T:210.5 /210.0 B:21.5 /0.0 T0:210.5 /210.0 @:63 B@:0 P:24.8 A:28.8
Recv: T:210.6 /210.0 B:21.6 /0.0 T0:210.6 /210.0 @:59 B@:0 P:24.7 A:28.6
Recv: T:211.7 /210.0 B:21.5 /0.0 T0:211.7 /210.0 @:40 B@:0 P:24.7 A:28.8
Recv: T:212.4 /210.0 B:21.4 /0.0 T0:212.4 /210.0 @:28 B@:0 P:24.8 A:29.0
Recv: T:213.6 /210.0 B:21.5 /0.0 T0:213.6 /210.0 @:8 B@:0 P:24.7 A:28.6
Recv: T:213.8 /210.0 B:21.4 /0.0 T0:213.8 /210.0 @:6 B@:0 P:24.8 A:28.8
Recv: T:213.3 /210.0 B:21.6 /0.0 T0:213.3 /210.0 @:13 B@:0 P:24.8 A:29.1
Recv: T:212.3 /210.0 B:21.8 /0.0 T0:212.3 /210.0 @:26 B@:0 P:24.8 A:29.0
Recv: T:211.6 /210.0 B:21.4 /0.0 T0:211.6 /210.0 @:35 B@:0 P:24.8 A:29.1
Recv: T:210.6 /210.0 B:21.5 /0.0 T0:210.6 /210.0 @:47 B@:0 P:24.9 A:29.1
Recv: T:210.6 /210.0 B:21.6 /0.0 T0:210.6 /210.0 @:42 B@:0 P:24.8 A:28.9
Recv: T:210.6 /210.0 B:21.4 /0.0 T0:210.6 /210.0 @:40 B@:0 P:24.8 A:29.0
Recv: T:211.1 /210.0 B:21.6 /0.0 T0:211.1 /210.0 @:28 B@:0 P:24.9 A:29.1
Recv: T:211.4 /210.0 B:21.6 /0.0 T0:211.4 /210.0 @:22 B@:0 P:24.9 A:29.1
Recv: T:211.7 /210.0 B:21.7 /0.0 T0:211.7 /210.0 @:16 B@:0 P:24.9 A:29.2

I trust this is the info you were after? Clearly setting the target temperatures to 0 for both bed and extruder after PID calibration is what is needed here in my opinion. This would hurt no one.

ruedli, 2021-08-22:
@iFrostizz 

And.... this leaves me to another observation, maybe storing the obtained PID values fails? I mean I get from the log 3 sets of PID, and instructions to store the last triple in the configuration file. When I do an M503, the reported PID values do not match the PID tuning.

[image]

Is there an easy explanation for this? Should another "issue" be opened for this observation?

iFrostizz, 2021-08-23:
Hi @ruedli,
I ran a PID autotune from the plugin, we can see the 5 cycles here:
[image]
And at the end of the PID, the constants are stored (in the volatile memory) and the temperature target will drop at 0 °C automatically.
[image]
I'm using Marlin 2.0.7.2, which may be part of the changes from Marlin 1.x.x which is the based version of Prusa firmware.

> The firmware for the Original Prusa i3 printers is proudly based on Marlin 1.0.x by Scott Lahteine

from https://github.com/prusa3d/Prusa-Firmware#prusa-firmware-mk3

As far as I can see, I haven't found the part containing the `#define Kp ...` where the constants are waiting for an `M500` to store them in the EEPROM on your screenshot.
I also found a tutorial on this link: https://www.snazzyit.com/pid-calibration-for-prusa-i3-mk2s-mk3-3d-printer/ explaining how to PID tune a MK2s/MK3 (might be including MK3s from your observations), and the constants are not stored there neither. We need to use `M301`.
So! I will probably also need to add the `M301 ...` command in the next update, as well as cooling down the extruder/bed before storing the data for Prusa users.

ruedli, 2021-08-23:
Yes, the fork for Prusa from Marlin is quite dated. Over that period, the two branches quite diverted, so what works in one branch is not necessarily working in the other. I just did a PID by issueing an M303, using U1. It definately does not store..  And you see: after the tuning your "target" temperature reported to octoprint is 0. Mine stays at the PID tuning temperature... I am still seeing 2 degrees of overshoot. Will reduce P a bit and see if that's better. Maybe the algorithm is flawed somehow.

iFrostizz, 2021-08-24:
@ruedli 
Maybe trying Marlin 2.0 would be a solution...
Anyway, thanks for that reporting this issue!

iFrostizz, 2021-08-30:
@ruedli
I merged the update some days ago in the "devel" branch if you wish to test it. If not, I will pull request into the main branch after adding more nozzle sizes in some days.

ruedli, 2021-08-31:
@iFrostizz 

Thank François,

I installed the 1.02 from the devel branch, but the temperature is reset. 

[image]

I ensured the mark was set for storing the PID parameters, however, I do not think they are actually stored. Please look at my output, I did an M503 afterwards. Also I think they are not stored? Af far as I can see you reset the fan speed, but not the temperature, I do not see an M104 S0. As soon as I manually provide that code, temperature goes down. Was that not supposed to be done in your devel branch?

Recv: T:44.2 /0.0 B:36.0 /0.0 T0:44.2 /0.0 @:0 B@:0 P:25.8 A:38.5
Recv: T:44.2 /0.0 B:36.0 /0.0 T0:44.2 /0.0 @:0 B@:0 P:25.8 A:38.2
Recv: T:44.0 /0.0 B:35.8 /0.0 T0:44.0 /0.0 @:0 B@:0 P:25.7 A:38.4
Recv: T:43.9 /0.0 B:35.9 /0.0 T0:43.9 /0.0 @:0 B@:0 P:25.8 A:38.4
Recv: T:43.8 /0.0 B:35.9 /0.0 T0:43.8 /0.0 @:0 B@:0 P:25.8 A:38.3
Recv: T:43.6 /0.0 B:35.9 /0.0 T0:43.6 /0.0 @:0 B@:0 P:25.8 A:38.0
Send: M106 S255
Recv: ok
Recv: T:43.6 /0.0 B:35.7 /0.0 T0:43.6 /0.0 @:0 B@:0 P:25.8 A:38.1
Send: M303 E0 C10 S215
Recv: PID Autotune start
Recv: T:43.33 @:127
Recv: T:44.81 @:127
Recv: T:49.02 @:127
Recv: T:54.14 @:127
Recv: T:59.53 @:127
Recv: T:65.43 @:127
Recv: T:71.72 @:127
Recv: T:77.01 @:127
Recv: T:83.05 @:127
Recv: T:88.21 @:127
Recv: T:93.97 @:127
Recv: T:99.53 @:127
Recv: T:104.54 @:127
Recv: T:109.95 @:127
Recv: T:115.38 @:127
Recv: T:119.84 @:127
Recv: T:125.09 @:127
Recv: T:129.38 @:127
Recv: T:134.44 @:127
Recv: T:139.06 @:127
Recv: T:143.49 @:127
Recv: T:148.24 @:127
Recv: T:152.94 @:127
Recv: T:156.91 @:127
Recv: T:161.18 @:127
Recv: T:165.20 @:127
Recv: T:169.30 @:127
Recv: T:173.70 @:127
Recv: T:177.41 @:127
Recv: T:181.45 @:127
Recv: T:185.77 @:127
Recv: T:189.20 @:127
Recv: T:193.16 @:127
Recv: T:196.88 @:127
Recv: T:199.93 @:127
Recv: T:204.15 @:127
Recv: T:207.50 @:127
Recv: T:210.99 @:127
Recv: T:214.74 @:127
Recv: T:217.66 @:0
Recv: T:219.84 @:0
Recv: T:220.25 @:0
Recv: T:219.11 @:0
Recv: T:217.29 @:0
Recv: T:214.90 @:127
Recv: T:212.34 @:127
Recv: T:210.73 @:127
Recv: T:211.30 @:127
Recv: T:213.39 @:127
[... 160 lines cut ...]
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E415.00
Recv: echo:UStep resolution:
Recv: echo: M350 X16 Y16 Z16 E16
Recv: echo:Maximum feedrates - normal (mm/s):
Recv: echo:  M203 X200.00 Y200.00 Z12.00 E120.00
Recv: echo:Maximum feedrates - stealth (mm/s):
Recv: echo:  M203 X100.00 Y100.00 Z12.00 E120.00
Recv: echo:Maximum acceleration - normal (mm/s2):
Recv: echo:  M201 X1000 Y1000 Z200 E5000
Recv: echo:Maximum acceleration - stealth (mm/s2):
Recv: echo:  M201 X960 Y960 Z200 E5000
Recv: echo:Acceleration: P=print, R=retract, T=travel
Recv: echo:  M204 P1000.00 R1250.00 T1000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B0.00 X8.00 Y8.00 Z0.40 E4.50
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! vvvvvvvvv !!!!!!!!!!!!!!!!!!!!!!

Recv: echo:PID settings:
Recv: echo:   M301 P20.50 I2.08 D50.42

[image]

iFrostizz, 2021-08-31:
@ruedli 
Sorry for that, I forgot to set the temperature at 0.
Did you activated the checkbox below the PID configuration to store the values automatically?

ruedli, 2021-08-31:
Yes, I did. I think the checkbox resulted in a M500 code sent, but not the
flag with the M301

iFrostizz, 2021-08-31:
@ruedli Okay, I know why.
Another "Marlin 1.x.x doesn't like marlin 2.x.x" because how I handled it was to wait for the line `"Recv: Kp: xx Ki: xx Kd:xx"`, and wait for the PID to actually finish to send the lastly found values. In your case, you are getting this instead: 

```
"Recv: Kp: xx"
"Recv: Ki: xx"
"Recv: Kd: xx"```

in separated lines. I will find a workaround

iFrostizz, 2021-08-31:
I hope that it is fixed now. If it worked, you cannot miss it... A message should appear 
[image]

ruedli, 2021-09-01:
@iFrostizz 

Hi François

Almost, but not there yet ;-)

Temperature goes down, there is an attempt to store the Pid values, but the values were undefined and consequently failed.

This is the log now, as well as the M503 output to verify what was stored: all zeroes.

Recv: T:88.8 /0.0 B:21.5 /0.0 T0:88.8 /0.0 @:0 B@:0 P:23.6 A:32.4
Recv: T:88.2 /0.0 B:21.4 /0.0 T0:88.2 /0.0 @:0 B@:0 P:23.6 A:32.3
Send: M303 E0 C5 S215
Recv: PID Autotune start
Recv: T:87.02 @:127
Recv: T:87.75 @:127
Recv: T:90.83 @:127
Recv: T:94.62 @:127
Recv: T:99.54 @:127
Recv: T:104.82 @:127
Recv: T:109.44 @:127
Recv: T:114.79 @:127
Recv: T:119.27 @:127
Recv: T:124.54 @:127
Recv: T:129.48 @:127
Recv: T:133.97 @:127
Recv: T:138.70 @:127
Recv: T:143.68 @:127
Recv: T:147.79 @:127
Recv: T:152.48 @:127
Recv: T:157.13 @:127
Recv: T:160.81 @:127
Recv: T:165.37 @:127
Recv: T:169.06 @:127
Recv: T:173.54 @:127
Recv: T:177.75 @:127
Recv: T:181.39 @:127
Recv: T:185.63 @:127
Recv: T:189.38 @:127
Recv: T:192.89 @:127
Recv: T:196.58 @:127
Recv: T:199.97 @:127
Recv: T:203.88 @:127
Recv: T:207.68 @:127
Recv: T:210.94 @:127
Recv: T:214.69 @:127
Recv: T:217.92 @:0
Recv: T:220.13 @:0
Recv: T:220.31 @:0
Recv: T:218.85 @:0
Recv: T:217.45 @:0
Recv: T:215.00 @:0
Recv: T:212.14 @:127
Recv: T:210.47 @:127
Recv: T:211.20 @:127
Recv: T:212.76 @:127
Recv: T:215.63 @:0
Recv: T:218.18 @:0
Recv: T:219.38 @:0
Recv: T:219.01 @:0
Recv: T:217.81 @:0
Recv: T:215.89 @:0
Recv:  bias: 116 d: 116 min: 210.36 max: 219.64
Recv: T:213.13 @:116
Recv: T:211.20 @:116
Recv: T:210.47 @:116
Recv: T:211.72 @:116
Recv: T:213.80 @:116
Recv: T:216.09 @:0
Recv: T:218.18 @:0
Recv: T:218.80 @:0
Recv: T:217.81 @:0
Recv: T:216.35 @:0
Recv:  bias: 117 d: 117 min: 210.47 max: 218.91
Recv: T:214.17 @:117
Recv: T:211.72 @:117
Recv: T:211.41 @:117
Recv: T:212.34 @:117
Recv: T:214.43 @:117
Recv: T:216.67 @:0
Recv: T:218.44 @:0
Recv: T:218.75 @:0
Recv: T:217.76 @:0
Recv: T:215.68 @:0
Recv:  bias: 113 d: 113 min: 211.20 max: 218.91
Recv:  Ku: 37.33 Tu: 19.67
Recv:  Classic PID
Recv:  Kp: 22.40
Recv:  Ki: 2.28
Recv:  Kd: 55.07
Recv: T:213.13 @:113
Recv: T:211.20 @:113
Recv: T:210.73 @:113
Recv: T:212.24 @:113
Recv: T:213.75 @:113
Recv: T:216.46 @:0
Recv: T:218.13 @:0
Recv: T:218.65 @:0
Recv: T:217.66 @:0
Recv: T:215.94 @:0
Recv:  bias: 114 d: 114 min: 210.63 max: 218.70
Recv:  Ku: 35.96 Tu: 20.20
Recv:  Classic PID
Recv:  Kp: 21.58
Recv:  Ki: 2.14
Recv:  Kd: 54.47
Recv: T:213.49 @:114
Recv: T:211.25 @:114
Recv: T:210.63 @:114
Recv: T:211.93 @:114
Recv: T:213.96 @:114
Recv: T:216.20 @:0
Recv: T:218.07 @:0
Recv: T:218.70 @:0
Recv: T:217.76 @:0
Recv: T:215.94 @:0
Recv:  bias: 115 d: 115 min: 210.63 max: 218.75
Recv:  Ku: 36.04 Tu: 20.19
Recv:  Classic PID
Recv:  Kp: 21.63
Recv:  Ki: 2.14
Recv:  Kd: 54.59
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h
Recv: ok
Recv: T:214.6 /215.0 B:21.5 /0.0 T0:214.6 /215.0 @:115 B@:0 P:24.1 A:33.9
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M106 S0
Recv: ok
Send: M301 E0 Pundefined Iundefined Dundefined
Recv: ok p:0.00 i:0.00 d:0.00 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M104 S0
Recv: ok
Send: M301 E0 Pundefined Iundefined Dundefined
Recv: ok p:0.00 i:0.00 d:0.00 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M104 S0
Recv: ok

Recv: T:212.4 /0.0 B:21.5 /0.0 T0:212.4 /0.0 @:0 B@:0 P:24.1 A:33.7
21.6 /0.0 T0:201.3 /0.0 @:0 B@:0 P:24.1 A:33.5
Recv: T:199.5 /0.0 B:21.6 /0.0 T0:199.5 /0.0 @:0 B@:0 P:24.1 A:33.8

Output from the M503:

Recv: echo:PID settings:
Recv: echo:   M301 P0.00 I0.00 D0.00

Conclusion: Almost, but not there yet ;-)

ruedli, 2021-09-01:
The message after calibration pops up, then disappears after a few seconds. As far as I could see, it did not have corect PID values though. Is that my settings, or do you intend it to popup and disappear that way?

iFrostizz, 2021-09-01:
Very probably an error in the code that is picking up the values from the received gcode commands. It is meant to fade out, but maybe that it makes more sense to let it "pinned" ?

iFrostizz, 2021-09-01:
@ruedli I hope that it's fine now..

ruedli, 2021-09-01:
I'll try again.. Going through some SW problems myself, trying to make my
OctoPlugout mqtt aware...

ruedli, 2021-09-01:
OK, think you nailed it now. And.. pinned dialog is better ;-)

Verified the settings, they were stored...

Nasty one, but glad we could work it out!

Cheers Ruud

Recv:  bias: 67 d: 67 min: 212.55 max: 217.34
Recv:  Ku: 35.61 Tu: 19.67
Recv:  Classic PID
Recv:  Kp: 21.36
Recv:  Ki: 2.17
Recv:  Kd: 52.53
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h
Recv: ok
Send: M500
Recv: T:214.9 /215.0 B:21.2 /0.0 T0:214.9 /215.0 @:67 B@:0 P:23.9 A:28.7
Recv: echo:Settings Stored
Recv: ok
Send: M106 S0
Recv: ok
Send: M301 E0 P21.36 I2.17 D52.53
Recv: ok p:21.36 i:2.17 d:52.53 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M104 S0

ruedli, 2021-09-01:
Ooopsss...

After the extruder tuning I did a (bed only) tuning. It ruined the PID settings for the extruder...

This seems OK: 

ecv: B:60.36 @:0
Recv: B:60.34 @:0
Recv: B:60.23 @:0
Recv: B:60.13 @:0
Recv:  bias: 57 d: 57 min: 59.44 max: 60.60
Recv:  Ku: 124.83 Tu: 48.26
Recv:  Classic PID
Recv:  Kp: 74.90
Recv:  Ki: 3.10
Recv:  Kd: 451.82
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h
Recv: ok
Recv: T:33.0 /0.0 B:59.9 /60.0 T0:33.0 /0.0 @:0 B@:57 P:27.0 A:38.1
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M301 E-1 P74.90 I3.10 D451.82
Recv: ok p:74.90 i:3.10 d:451.82 c:1.00
Send: M500
Recv: ok
Send: M140 S0
Recv: echo:Settings Stored
Recv: ok

But when I check my settings with M503 I get:

ecv: echo:PID settings:
Recv: echo:   M301 P74.90 I3.10 D451.82
Recv: echo:PID heatbed settings:
Recv: echo:   M304 P88.01 I3.90 D496.19

Not sure how the extruder ones got changed... Did not do anything manually. Can you reproduce this?

[image]

ruedli, 2021-09-01:
The OTHER window in which I had octoprint open (not on the calibration tab) reported this:

[image]

No idea where it picked this up, but this is where the wrong settings came from after bed-only-tuning

iFrostizz, 2021-09-01:
Well, it looks like I'm sending `M301` for the bed and for the extruder too. And the gcode to send is `M304`, the index for extruders is between 0 to ...

ruedli, 2021-09-01:
I am now issuing both extruder/bed tuning. The be tuning finished. On the window of the calibration I have:

[image]

But on the OTHER window I have:

[image]

That's a bit strange isn't it?

iFrostizz, 2021-09-01:
It is, could I see your logs please ?

ruedli, 2021-09-01:
ecv: B:60.53 @:0
Recv: B:60.45 @:0
Recv: B:60.34 @:0
Recv: B:60.23 @:0
Recv: B:60.14 @:0
Recv: B:60.00 @:0
Recv:  bias: 55 d: 55 min: 59.48 max: 60.54
Recv:  Ku: 132.30 Tu: 45.37
Recv:  Classic PID
Recv:  Kp: 79.38
Recv:  Ki: 3.50
Recv:  Kd: 450.17
Recv: B:59.91 @:55
Recv: B:59.80 @:55
Recv: B:59.63 @:55
Recv: B:59.58 @:55
Recv: B:59.55 @:55
Recv: B:59.48 @:55
Recv: B:59.52 @:55
Recv: B:59.53 @:55
Recv: B:59.69 @:55
Recv: B:59.70 @:55
Recv: B:59.81 @:55
Recv: B:59.89 @:55
Recv: B:60.14 @:0
Recv: B:60.25 @:0
Recv: B:60.31 @:0
Recv: B:60.57 @:0
Recv: B:60.54 @:0
Recv: B:60.54 @:0
Recv: B:60.54 @:0
Recv: B:60.48 @:0
Recv: B:60.38 @:0
Recv: B:60.16 @:0
Recv: B:60.16 @:0
Recv: B:60.06 @:0
Recv:  bias: 56 d: 56 min: 59.47 max: 60.63
Recv:  Ku: 122.92 Tu: 47.73
Recv:  Classic PID
Recv:  Kp: 73.75
Recv:  Ki: 3.09
Recv:  Kd: 439.98
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h
Recv: ok
Send: M500
Recv: T:23.8 /0.0 B:60.0 /60.0 T0:23.8 /0.0 @:0 B@:56 P:26.1 A:39.3
Recv: echo:Settings Stored
Recv: ok
Send: M303 E0 C5 S215
Recv: PID Autotune start
Recv: T:23.72 @:127
Recv: T:25.75 @:127
Recv: T:30.73 @:127
Recv: T:35.50 @:127
Recv: T:42.00 @:127
Recv: T:47.61 @:127
Recv: T:54.00 @:127
Recv: T:60.67 @:127
Recv: T:66.07 @:127
Recv: T:72.39 @:127
Recv: T:78.54 @:127
Recv: T:83.82 @:127
Recv: T:89.70 @:127
Recv: T:95.42 @:127
Recv: T:100.31 @:127
Recv: T:105.98 @:127
Recv: T:110.74 @:127
Recv: T:116.10 @:127
Recv: T:121.33 @:127
Recv: T:125.89 @:127
Recv: T:130.88 @:127
Recv: T:135.94 @:127
Recv: T:139.99 @:127
Recv: T:145.13 @:127
Recv: T:149.73 @:127
Recv: T:153.96 @:127
Recv: T:158.41 @:127
Recv: T:162.43 @:127
Recv: T:166.76 @:127
Recv: T:171.00 @:127
Recv: T:174.81 @:127
Recv: T:178.96 @:127
Recv: T:183.21 @:127
Recv: T:186.82 @:127
Recv: T:190.63 @:127
Recv: T:194.51 @:127
Recv: T:197.60 @:127
Recv: T:201.61 @:127
Recv: T:205.00 @:127
Recv: T:208.66 @:127
Recv: T:212.60 @:127
Recv: T:215.89 @:0
Recv: T:218.59 @:0
Recv: T:220.50 @:0
Recv: T:219.90 @:0
Recv: T:218.28 @:0
Recv: T:216.51 @:0
Recv: T:213.96 @:127
Recv: T:211.35 @:127
Recv: T:210.89 @:127
Recv: T:212.34 @:127
Recv: T:214.84 @:127
Recv: T:217.34 @:0
Recv: T:219.17 @:0
Recv: T:219.22 @:0
Recv: T:218.07 @:0
Recv: T:216.51 @:0
Recv:  bias: 116 d: 116 min: 210.89 max: 219.53
Recv: T:213.96 @:116
Recv: T:211.46 @:116
Recv: T:210.73 @:116
Recv: T:211.41 @:116
Recv: T:213.80 @:116
Recv: T:215.94 @:0
Recv: T:217.97 @:0
Recv: T:218.39 @:0
Recv: T:217.76 @:0
Recv: T:215.78 @:0
Recv:  bias: 116 d: 116 min: 210.73 max: 218.65
Recv: T:213.49 @:116
Recv: T:211.72 @:116
Recv: T:211.51 @:116
Recv: T:212.81 @:116
Recv: T:214.74 @:116
Recv: T:217.03 @:0
Recv: T:218.70 @:0
Recv: T:218.65 @:0
Recv: T:217.45 @:0
Recv: T:215.31 @:0
Recv:  bias: 113 d: 113 min: 211.15 max: 218.91
Recv:  Ku: 37.08 Tu: 19.92
Recv:  Classic PID
Recv:  Kp: 22.25
Recv:  Ki: 2.23
Recv:  Kd: 55.41
Recv: T:213.33 @:113
Recv: T:211.35 @:113
Recv: T:211.04 @:113
Recv: T:212.34 @:113
Recv: T:214.38 @:113
Recv: T:216.61 @:0
Recv: T:218.49 @:0
Recv: T:218.85 @:0
Recv: T:218.02 @:0
Recv: T:215.89 @:0
Recv:  bias: 114 d: 114 min: 210.94 max: 218.85
Recv:  Ku: 36.67 Tu: 20.20
Recv:  Classic PID
Recv:  Kp: 22.00
Recv:  Ki: 2.18
Recv:  Kd: 55.54
Recv: T:213.59 @:114
Recv: T:211.51 @:114
Recv: T:211.09 @:114
Recv: T:212.19 @:114
Recv: T:214.53 @:114
Recv: T:216.93 @:0
Recv: T:218.65 @:0
Recv: T:218.96 @:0
Recv: T:218.02 @:0
Recv: T:216.46 @:0
Recv:  bias: 113 d: 113 min: 210.99 max: 219.06
Recv:  Ku: 35.64 Tu: 20.74
Recv:  Classic PID
Recv:  Kp: 21.39
Recv:  Ki: 2.06
Recv:  Kd: 55.44
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h
Recv: ok
Recv: T:214.5 /215.0 B:45.9 /0.0 T0:214.5 /215.0 @:113 B@:0 P:26.4 A:37.7
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M106 S0
Recv: ok
Send: M301 E0 P73.75 I3.09 D439.98
Recv: ok p:73.75 i:3.09 d:439.98 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M104 S0
Recv: ok
Send: M301 E-1 P73.75 I3.09 D439.98
Recv: ok p:73.75 i:3.09 d:439.98 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M140 S0
Recv: ok
Send: M301 E0 P21.39 I2.06 D55.44
Recv: ok p:21.39 i:2.06 d:55.44 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M104 S0
Recv: ok
Send: M301 E-1 P21.39 I2.06 D55.44
Recv: ok p:21.39 i:2.06 d:55.44 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M140 S0
Recv: ok
Recv: T:211.9 /0.0 B:45.5 /0.0 T0:211.9 /0.0 @:0 B@:0 P:26.3 A:37.6
Recv: T:209.6 /0.0 B:45.5 /0.0 T0:209.6 /0.0 @:0 B@:0 P:26.4 A:37.7
Recv: T:208.0 /0.0 B:45.4 /0.0 T0:208.0 /0.0 @:0 B@:0 P:26.4 A:37.4
Recv: T:206.3 /0.0 B:45.3 /0.0 T0:206.3 /0.0 @:0 B@:0 P:26.3 A:37.6
Recv: T:204.3 /0.0 B:45.1 /0.0 T0:204.3 /0.0 @:0 B@:0 P:26.5 A:37.3
Recv: T:202.6 /0.0 B:45.0 /0.0 T0:202.6 /0.0 @:0 B@:0 P:26.4 A:37.3
Recv: T:200.7 /0.0 B:44.9 /0.0 T0:200.7 /0.0 @:0 B@:0 P:26.4 A:37.3
Recv: T:199.2 /0.0 B:44.7 /0.0 T0:199.2 /0.0 @:0 B@:0 P:26.4 A:37.4
Recv: T:197.8 /0.0 B:44.8 /0.0 T0:197.8 /0.0 @:0 B@:0 P:26.4 A:37.4
Recv: T:196.4 /0.0 B:44.5 /0.0 T0:196.4 /0.0 @:0 B@:0 P:26.4 A:37.4

ruedli, 2021-09-01:
This one seems wrong...

Send: M301 E0 P73.75 I3.09 D439.98

iFrostizz, 2021-09-01:
I see... The PID applying command should be sent just after that the PID is finished.
Here, the buffer is full and the command is queued.

ruedli, 2021-09-01:
I'll wait for another version to try, but this is still a problem I guess. Don't rush, make sure it works at your place first ;-)

ruedli, 2021-09-01:
Some last info:

M503 reports the stored settings (after combined bed/extruder pid calibration)

[image]

Alltrhough they seem ok-ish, the "bed" figures appear nowhere... And in between some extruder 0 settings occur that are wiped out by the last extruder PID.

[image]

[image]

iFrostizz, 2021-09-01:
So I think that I will just let the user do one PID at a time by disabling the other textbox, that will make it more "bug-proof"

ruedli, 2021-09-01:
Not really, since when I did them one after the other, I STILL got the extruder PID values written to the bed.

iFrostizz, 2021-09-01:
Did you tried the version sending an `M304` for the bed ?

ruedli, 2021-09-01:
Did you update the version number so I can check?

ruedli, 2021-09-01:
Reinstalled and restarted octoprint: new test.

iFrostizz, 2021-09-01:
I didn't implemented the greying out input box yet, I will do it tomorrow

ruedli, 2021-09-01:
Retested, same problems.

[image]

[image]

Recv: B:60.14 @:0
Recv: B:60.11 @:0
Recv:  bias: 57 d: 57 min: 59.38 max: 60.56
Recv: B:59.91 @:57
Recv: B:59.78 @:57
Recv: B:59.67 @:57
Recv: B:59.64 @:57
Recv: B:59.50 @:57
Recv: B:59.50 @:57
Recv: B:59.47 @:57
Recv: B:59.56 @:57
Recv: B:59.59 @:57
Recv: B:59.70 @:57
Recv: B:59.83 @:57
Recv: B:60.00 @:57
Recv: B:60.08 @:0
Recv: B:60.20 @:0
Recv: B:60.41 @:0
Recv: B:60.51 @:0
Recv: B:60.54 @:0
Recv: B:60.64 @:0
Recv: B:60.53 @:0
Recv: B:60.51 @:0
Recv: B:60.45 @:0
Recv: B:60.30 @:0
Recv: B:60.22 @:0
Recv: B:60.09 @:0
Recv:  bias: 57 d: 57 min: 59.47 max: 60.64
Recv:  Ku: 123.59 Tu: 46.69
Recv:  Classic PID
Recv:  Kp: 74.15
Recv:  Ki: 3.18
Recv:  Kd: 432.77
Recv: B:59.98 @:57
Recv: B:59.80 @:57
Recv: B:59.69 @:57
Recv: B:59.61 @:57
Recv: B:59.59 @:57
Recv: B:59.58 @:57
Recv: B:59.58 @:57
Recv: B:59.70 @:57
Recv: B:59.83 @:57
Recv: B:59.92 @:57
Recv: B:60.06 @:0
Recv: B:60.14 @:0
Recv: B:60.23 @:0
Recv: B:60.42 @:0
Recv: B:60.50 @:0
Recv: B:60.50 @:0
Recv: B:60.45 @:0
Recv: B:60.45 @:0
Recv: B:60.39 @:0
Recv: B:60.25 @:0
Recv: B:60.16 @:0
Recv: B:60.09 @:0
Recv:  bias: 54 d: 54 min: 59.55 max: 60.57
Recv:  Ku: 134.19 Tu: 45.11
Recv:  Classic PID
Recv:  Kp: 80.51
Recv:  Ki: 3.57
Recv:  Kd: 453.96
Recv: B:59.98 @:54
Recv: B:59.81 @:54
Recv: B:59.64 @:54
Recv: B:59.55 @:54
Recv: B:59.53 @:54
Recv: B:59.47 @:54
Recv: B:59.47 @:54
Recv: B:59.50 @:54
Recv: B:59.59 @:54
Recv: B:59.67 @:54
Recv: B:59.77 @:54
Recv: B:59.88 @:54
Recv: B:59.95 @:54
Recv: B:60.11 @:0
Recv: B:60.23 @:0
Recv: B:60.36 @:0
Recv: B:60.45 @:0
Recv: B:60.50 @:0
Recv: B:60.53 @:0
Recv: B:60.50 @:0
Recv: B:60.34 @:0
Recv: B:60.31 @:0
Recv: B:60.14 @:0
Recv: B:60.05 @:0
Recv:  bias: 57 d: 57 min: 59.42 max: 60.56
Recv:  Ku: 127.84 Tu: 47.23
Recv:  Classic PID
Recv:  Kp: 76.70
Recv:  Ki: 3.25
Recv:  Kd: 452.85
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from above into Configuration.h
Recv: ok
Recv: T:61.0 /0.0 B:60.0 /60.0 T0:61.0 /0.0 @:0 B@:57 P:26.3 A:40.3
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M106 S255
Recv: ok
Send: M117 OP: OctoPlugout 3.1
Recv: LCD status changed
Recv: ok
Send: M304 P20.15 I2.00 D50.86
Recv: ok p:20.15 i:2.00 d:50.86
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M140 S0
Recv: ok
Send: M301 E0 P20.15 I2.00 D50.86
Recv: ok p:20.15 i:2.00 d:50.86 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M104 S0
Recv: ok
Send: M304 P76.70 I3.25 D452.85
Recv: ok p:76.70 i:3.25 d:452.85
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M140 S0
Recv: ok
Send: M301 E0 P76.70 I3.25 D452.85
Recv: ok p:76.70 i:3.25 d:452.85 c:1.00
Recv: ok
Send: M500
Recv: echo:Settings Stored
Recv: ok
Send: M104 S0
Recv: ok
Recv: T:60.7 /0.0 B:59.9 /0.0 T0:60.7 /0.0 @:0 B@:0 P:26.4 A:40.2
Recv: T:60.3 /0.0 B:59.7 /0.0 T0:60.3 /0.0 @:0 B@:0 P:26.4 A:40.2
Recv: T:59.8 /0.0 B:59.6 /0.0 T0:59.8 /0.0 @:0 B@:0 P:26.4 A:40.0
Recv: T:59.3 /0.0 B:59.4 /0.0 T0:59.3 /0.0 @:0 B@:0 P:26.4 A:40.1
Recv: T:58.7 /0.0 B:59.3 /0.0 T0:58.7 /0.0 @:0 B@:0 P:26.4 A:40.0
Recv: T:58.1 /0.0 B:59.2 /0.0 T0:58.1 /0.0 @:0 B@:0 P:26.4 A:39.9

iFrostizz, 2021-09-01:
Yes, you ran the PID for both bed and hotend at the same time, right?
I meant when you do it separately. I will just modify the UI to prevent the user from entering both values at the same time.

ruedli, 2021-09-02:
Well, change it, test it and when it works for you, publish it and I will then test it as well. Note that it could be an issue that is shown only when there are two windows open. I have that and both show _**different**_ messages, so there is a difference in behavior that should not happen and could result in problems for other users with different configurations and different timing.

iFrostizz, 2021-09-02:
I just pushed the new version, with the disabled input to prevent the user from typing in the two boxes at the same time.
[image]
[image]
And it looks like I got the same results from the two octoprint instances.
Sadly, I cannot try the bed pid because it is not enabled in my firmware, and again I don't have Marlin 1.x.x

ruedli, 2021-09-02:
OK, I will have a go, doing extruder and the bed PID calibration with this.

ruedli, 2021-09-02:
The key to the solution could be that you ONLY execute pop-ups and M301 / M304 settings for the window where the PID tuning is executed from. Then I feel it should work, as that window picks up the right extruder... The good news is that you can test that functionality as well without capability to run bed PID tuning.

Is there a way in your script that does the popup and M301/M304 to be aware of the client context and whether autoleveling is running?

ruedli, 2021-09-02:
And.... with that client / calibration PID tuning awareness you can execute both in one run again, no need to make them mutually exclusive.....

iFrostizz, 2021-09-05:
Hey @ruedli .
I just enabled the bed PID tuning on my printer, ran a PID from the plugin, and opened several windows.
I got the same issue as you and finally figured out why that would happen, I was confused!
So now, the notification should not spawn anymore nor the wrong settings be applied on a window that didn't started a PID tuning. I think that I will also grey out the button that runs the tuning for all sessions once it has been clicked and until a PID was finished because that would induce the same bug if pressed from another session. I will do as well for the progress bar (show the same percentage for every sessions)

ruedli, 2021-09-06:
@iFrostizz  That is good to hear . Now that you can reproduce it, I am sure you can get it fixed. The multiple window was probably something not considered in the original design and was causing errors and confusion. Maybe in more places? Best to close this issue when you release a version with fixes it and document the version in which this is fixed.

skl111, 2021-12-21:
Anyone know the status of this issue?

ruedli, 2021-12-21:
I have tested versions from the devel branch and many problems were resolved. It also helped not having multiple octoprint windows open. I left it to @iFrostizz to close this issue when it was merged to main.

iFrostizz, 2021-12-21:
Hey @skl111 and @ruedli , sorry for the lack of news but I'm quite busy currently. For now I'd suggest to use the devel branch and I'll add more features to it and PR to main when this one will be satisfying. Thanks for helping ruedli!

skl111, 2021-12-21:
Will give it a go, thanks!

skl111, 2022-02-11:
I tried the devel branch and it will let you input the values, but after you do that and hit "Run PID Autotune" it does nothing, nothing gets sent to the printer.