franfrandev/OctoPrint-CalibrationCompanion #32: Generated gcodes do not set any temperature so it tries to print cold

kdkwarhead, 2021-11-22:
I don’t know why this is happening to me but the generated gcodes do not set any temperature so it tries to print cold. There are no M140 or M104 commands in the gcode.
It works well if I manually add temperature gcodes or if I add it to "start Gcode" in CalibrationCompanion.
I tried Flow, Retraction and Temperature with the same result.
I have the Smart Preheat plugin installed but I get the same result if I deactivate it.

**I entered the following Start Gcode:**
M301 P22.6 I2.0 D63.4; Set PID for hotend
M205 E12 ;Jerk extruder
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
M92 E431 ; Estep
M900 K0.13 ;Pressure advance
G28 ;Home
G92 E0 ;Reset Extruder
G0 Z2.0 F3000             ;Move Z Axis up
G0 X5   Y3 Z2.0 F2500     ;Move to start position
G0 X5   Y3 Z0.4 F2500     ;Move to start position
G1 X145 Y3 Z0.4 F1000 E20 ;Draw the first line
G0 X160 Y3 Z0.4 F500      ;Draw a short line without extrusion (swiping)
G92 E0                    ;Reset Extruder
G1 Z2.0 F2500             ;Move Z Axis up

[octoprint-systeminfo-20211122164420.zip](https://github.com/iFrostizz/OctoPrint-CalibrationCompanion/files/7582494/octoprint-systeminfo-20211122164420.zip)

iFrostizz, 2021-11-22:
Hey, that's happening because you need to set the temperature variable in the G-code textbox manually e.g: 
```
M104 S[regular_nozzle];set hotend to [regular_nozzle]C
M109 S[regular_nozzle];wait hotend to [regular_nozzle]C
```
These variables will be replaced by the one you entered in the print settings.
Please check this part of the README: https://github.com/iFrostizz/OctoPrint-CalibrationCompanion#variables
I also pre-saved a profile for you, it's stored in the profile1 by default and you can load it from the profiles tab.

kdkwarhead, 2021-11-23:
You are right. I didn’t read the manual properly. It worked when I added the following:
M140 S[regular_bed];set hotbed to [regular_bed]C
M190 S[regular_bed]; wait hotbed to [regular_bed]C
M104 S[regular_nozzle];set hotend to [regular_nozzle]C
M109 S[regular_nozzle];wait hotend to [regular_nozzle]C
I could only print 3 stages then the part come of my build plate since the base area is quite small. Its ok for me but a brim would be nice.

Thank you.

closed 2021-11-23 by kdkwarhead